wokwi/wokwi-features #5: possibility to edit the project - `Later`

Puneethkumarsn, 2020-10-22:
I came up with the need to edit the project I had already shared with many. A provision for the `original author` to update the project is preferred. Seems like logging-in is the only option I believe

urish, 2020-10-22:
Agreed! For now, as long as you use the same browser, you should have access to edit the project. You can find a list of all the projects you have access edit to on https://wokwi.com/arduino/projects

Puneethkumarsn, 2020-10-22:
This works for me for now

urish, 2020-12-21:
There's now also an option to sign in, using either a Google or a GitHub account:

[image]

closed 2020-12-21 by urish