wokwi/wokwi-features #15: diagram.jSON file fails to load for the second project

Puneethkumarsn, 2020-10-23:
Steps followed:

1. Open https://wokwi.com/arduino/libraries/FastLED/DigitalRain and play with the simulation.
2. now, go back (you will go to FastLED example page)
3. Open Fire example now. Wait until the page has completed loading
4. Do not run the code yet
5. Open JSON file and try to edit UNO pin number randomly (but, leave it as is finally)
6. Run the code

**Observation 1:**

- `Fire example` runs on `Digital rain` element

- `Digital rain` example runs on `Fire example`

Below pictures for reference:
[image]
[image]

Some more pics...
[image]
[image]

**Observation 2:**
The critical thing is step number 5 above. If one executes the code once, before editing any part of the code, the INO file and JSON file will be up to date in the right way. if not, there is some possibility, where the diagram.JSON file will stop loading for the newly opened project. 

ignore the text below:
I do not know the background on how it works but as of now, the old JSON file seems to be there until the new one replaces it (when I open a second project). The way how the newly opened JSON file replaces the old one seems weak.

urish, 2020-11-08:
Thanks for reporting! I believe that I found the bug and fixed it, can you please check and confirm?

Puneethkumarsn, 2020-11-08:
I too think so. I could not recreate it now 😃

closed 2020-11-08 by Puneethkumarsn