wokwi/wokwi-features #8: The timer and the performance meter doesn't come in view

Puneethkumarsn, 2020-10-23:
Earlier, when the window size was reduced, the timer and the performance meter used to come in the available window, so that user can still see them without scrolling to the extreme right.
Now, it is missing. I might be wrong too.

https://wokwi.com/playground/neopixel-strip

[image]

earlier reference:

[image]

urish, 2020-10-23:
Workaround until this is fixed:
1. Resize the window
2. Refresh the page
3. Start the simulation

urish, 2020-10-23:
Fixed!

closed 2020-10-23 by urish

Puneethkumarsn, 2020-10-23:
it's perfect

Gavin-Perry, 2023-11-26:
https://wokwi.com/playground/neopixel-strip in fact the whole playground is not available (links to home page). What happened?

urish, 2023-11-26:
We deprecated the playgrounds two years ago: https://github.com/wokwi/wokwi-playgrounds

Gavin-Perry, 2023-11-28:
Thanks for letting me know. I'm new to wokwi.
Well I have your attention the other thing that is odd is that the new pixel strip isn't available from the + add components menu. Only the ring and singe LED are available. I had to copy it from another project I found online into the .json file and figure out how to edit it.

Finally, I have two strips that are sent the same data and yet colors come out completely different depending on the brightness level I set. One saturates to white the other saturates to black and neither shows the colors as they are in real life.

Gavin

Sent via the Samsung Galaxy S22 5G, an AT&T 5G smartphone