wokwi/wokwi-features #18: The simulation window goes blank + The comments are not formatted

Puneethkumarsn, 2020-11-01:
I was playing with Arduino Mega simulation 

https://wokwi.com/arduino/libraries/demo/blink-mega

1. Changed the Mega to Nano (in the JSON file)
2. Simulation window updated to show Arduino Nano
3. Clicked on the blink-mega.ino file
4. The window in the right (simulation window) went blank (black)
below are the images

[image]
[image]

_________________________
Checks

1. The simulation window will also go blank if I miss punctuation. I am sure, I did not miss at that time (99%)
2. Even if I had missed it accidentally, the comments in the *.ino file won't be marked in red underlines.
3. to try out, I simply, made a punctuation mistake intentionally, and here is the screenshot (showing the comments being formatted properly) 
[image]

[image]

urish, 2020-11-03:
Hi, I deployed a fix (also for #15) - can you please test again and report if the issue is gone?

Puneethkumarsn, 2020-11-06:
it works :)

closed 2020-11-06 by Puneethkumarsn