wokwi/wokwi-features #24: Feature Request

Puneethkumarsn, 2020-11-28:
Link reference: https://github.com/FastLED/FastLED/issues/1076

The request is below:

> 
> Some people make matrix-based table lamps and screens. Could you add a frozen glass (light scattering) effect on top of the matrix? It would help to see the light effect live, like on a lamp.
> 
> Our project in progress https://github.com/DmytroKorniienko/FireLamp_JeeUI/tree/dev
>

urish, 2020-11-29:
Hello @kostyamat, can you please give a photo or more details of what you want to effect to look like?

We currently have an ability to blur. Compare this (pixelated):

[image]

vs this (blurry):

[image]

(demo from [this page](https://wokwi.com/arduino/libraries/FastLED/DigitalRain))

kostyamat, 2020-11-30:
Hi. 
For example reworked "Fireworks" https://youtu.be/V-VqwKp6tXE, or any other on my YouTube channel.

Puneethkumarsn, 2021-05-17:
voting to close this. Blue feature is sufficient and address 99% cases where a blur is needed

closed 2021-05-17 by urish