wokwi/wokwi-features #25: Simulation stops at 1.5 seconds

Puneethkumarsn, 2020-11-30:
**Link**: 
https://wokwi.com/arduino/projects/283628426466689548

Once the timer reaches 1.5 seconds, it will be stuck. refresh and retry did not solve.

1. Took colour palette example from https://wokwi.com/arduino/libraries/FastLED/ColorPalette
2. Change the Arduino type to ATtiny85 (wokwi-attiny85)
3. update the pin numbers
4. Run

urish, 2020-12-01:
Issue reproduced and confirmed - I think I know what's causing it. Expect a fix soon!

urish, 2020-12-01:
Issue fixed - code doesn't stop at 1.5 seconds. Also fixed the simulation speed (so `delay(3000)` actually does 3 second delay and not 1.5).

Note that the FastLED example still doesn't work - there's some issue with the timing, so I'll keep this open as I'm investigating the timing issue

closed 2020-12-01 by urish

Puneethkumarsn, 2020-12-01:
That's is good news