wokwi/wokwi-features #26: hard Refresh also doesn't load the complete project

Puneethkumarsn, 2020-12-04:
**Link**: https://wokwi.com/arduino/projects/283991385282970124

1. After doing minor changes in the digram.JSON file I saved it
2. I now clicked on the save button
3. Now, if I reload the page, all my changes were gone
4. I also tried 'Ctrl' + 'Shift' + 'R'
5. If I open the same link in Stealth mode or on another browser, I can see all my latest changes

[image]

Puneethkumarsn, 2020-12-04:
**Second update:**

1. I saved the design by clicking on the 'Save' button
2. And did `Ctrl` + `Shift` + `R`
3. Diagram.JSON file lost all the changes and loaded the old status/random old status contents

[image]

urish, 2020-12-04:
Thanks for reporting! I deployed a fix - it might need to refresh twice, but it should hopefully improve the situation. Can you please try and let me know?

Puneethkumarsn, 2020-12-05:
It looks okay now 👍 I don't know why but the update indeed happened only with the second hard refresh.

urish, 2020-12-05:
Yes, it happens because of stale cache. I hope to fix that in the future as well...

closed 2020-12-05 by urish