wokwi/wokwi-features #29: the simulation fails for an edit - works in stealth mode though

Puneethkumarsn, 2020-12-13:
1. open the link: https://wokwi.com/arduino/projects/279890929738842636
2. clear the editor window and paste the below code

```
//www.youtube.com/Electronics is Fun//
//www.facebook.com/Electronics is Fun//
//Project by MOHD SOHAIL//

int pinsCount=10;                        // declaring the integer variable pinsCount
int pins[] = {2,3,4,5,6,7,8,9,10,11};          // declaring the array pins[]
 
void setup() {                
  for (int i=0; i<pinsCount; i=i+1){    // counting the variable i from 0 to 9
    pinMode(pins[i], OUTPUT);            // initialising the pin at index i of the array of pins as OUTPUT
  }
}
 
void loop() {
  for (int i=0; i<pinsCount; i=i+1){    // chasing right
    digitalWrite(pins[i], HIGH);         // switching the LED at index i on
    delay(100);                          // stopping the program for 100 milliseconds
    digitalWrite(pins[i], LOW);          // switching the LED at index i off
  }
  for (int i=pinsCount-1; i>0; i=i-1){   // chasing left (except the outer leds)
    digitalWrite(pins[i], HIGH);         // switching the LED at index i on
    delay(100);                          // stopping the program for 100 milliseconds
    digitalWrite(pins[i], LOW);          // switching the LED at index i off
  }
}
```

3. in my PC, if I try to compile, it fails. Also noticed is that, when I save teh modified code, the new number is not assigned (one hint)

4. Now, if I take the link (in step 1) and open it in the stealth window, it works as expected

[image]

[image]

urish, 2020-12-13:
Thanks for reporting! Can you please open the browser devtools console and see what it says there?

Puneethkumarsn, 2020-12-13:
[image]

interestingly, in the same browser, the old tab still reports failure. I opened the same link in a new tab--> it works. The older tab still fails to compile.

urish, 2020-12-13:
According to the error message, it seems like there's no compilation error, rather than an issue connecting to the build server (ERR_CONNECTION_TIMED_OUT). I'm leaving this issue open as a reminder to show a better error message when this happens.

urish, 2020-12-13:
Error message updated!

closed 2020-12-13 by urish

Puneethkumarsn, 2020-12-13:
I saw the update.

one more observation: 
The error pops up when the RUN button is clicked without hitting the save button after an edit.
If I save the code before RUNning, the simulation works fine

urish, 2020-12-13:
Can you take a look at the "Network" tab on the browser devtools while doing it? It may give more information why the connection fails...

Puneethkumarsn, 2020-12-13:
unable to create the error now. Will monitor the Network tab when the error appears again

Puneethkumarsn, 2020-12-13:
[image]