wokwi/wokwi-features #31: feature request - how to make a project Readonly

Puneethkumarsn, 2020-12-13:
is there a possibility to make the project read only?

can we make it as an option or a feature?

background: 

I had created a project and had used it in some of the articles already online.

When I had to create a new project, I used the same project and updated the contents of the editor and hit SAVE button.

I was anticipating a new number (new link) for the new project, but (maybe due to the same PC, browser)

urish, 2020-12-13:
Great suggestion! This is now available under the Save menu:

[image]

When "Lock project" is enabled, the "Save" option is automatically disabled, and it's only possible to "Save a copy" until you disable "Lock project".

closed 2020-12-13 by urish

Puneethkumarsn, 2020-12-13:
wow that was fast