wokwi/wokwi-features #32: Graphical way to select Arduino elements and connect them with the Arduino

Puneethkumarsn, 2020-12-24:
We presently have 

https://wokwi.com/arduino/alpha-editor

New feature request is to make it 
1. graphical and easier in terms of drawing wires between the pins.
2. select Arduino UNO, Arduino Mega, Arduino Nano, ATtiny85 (and more when the new items are added)
3. select all the Wokwi elements (peripherals)
4. draw by connecting wires to choose the colours

stevesigma, 2021-01-14:
I don't think so many requests in one is good to read what is done and what is unfinished.
1. and 4. are close and should be done in one shot.

urish, 2021-02-08:
We now have the ability to drag components.

So I guess adding elements by selecting them from a list / drawing wires with the mouse will come next :)

urish, 2021-05-06:
Mostly done! 

https://docs.wokwi.com/guides/diagram-editor

closed 2021-05-06 by urish