wokwi/wokwi-features #36: zooming the browser and the simulation window

Puneethkumarsn, 2021-01-03:
**Summary**:
if I use two-finger zoon to zoom the area of the simulation window, 

1. The whole browser zooms out or zooms in. I cannot zoom the simulation window alone (Firefox browser)
2. it works as expected (Avast secure browser)

**Video** link:
https://drive.google.com/file/d/15FornSn2EV0cVLg-4z2f1Z39t4jmp_gd/view?usp=sharing

Puneethkumarsn, 2021-05-17:
Two-finger zoom works on firefox now. was there any fix done? can we close this issue? I will reopen if I face a problem in the future. I am using only firefox now.

urish, 2021-05-17:
Nothing done specifically to address this, but we did have some changes in the diagram editor recently. So... who knows? ;-)

closed 2021-05-17 by urish