wokwi/wokwi-features #38: il9163 program fails to compile

Puneethkumarsn, 2021-01-03:
Project:
https://wokwi.com/arduino/libraries/lcdgfx/il9163_demo

Error:
[image]

Error text:
'class DisplayIL9163_128x128x16_SPI' has no member named 'negativeMode'

/home/hexi/Arduino/libraries/lcdgfx/src/v2/lcd/base/ssd1306_common.inl:702: 'class NanoDisplayOps<NanoDisplayOps16<InterfaceIL9163<PlatformSpi> >, InterfaceIL9163<PlatformSpi> >' has no member named 'positiveMode'

urish, 2021-01-03:
Seems like the library code has recently changed:

https://github.com/lexus2k/lcdgfx/blob/master/examples/demos/il9163_demo/il9163_demo.ino

urish, 2021-01-03:
Thanks for reporting, fixed!

closed 2021-01-03 by urish