wokwi/wokwi-features #39: new request - all project index

Puneethkumarsn, 2021-01-07:
All project index based on projects from all the users across the globe!

**Features**:

The project list contains all the projects created on wokwi site. The filename of the project seldom tells what exactly it involves. so, for the purpose of filtering, we can have 

1. Project filtering based on the wokwi elements used in it
2. Project filtering based on the number of elements used (complexity)
3. Project based on likes from users / popularities
4. Any other ideas goes here

stevesigma, 2021-01-14:
This view should be similar to user's projects page. (sorting, columns) with minor differences.

urish, 2021-01-14:
Updates:

- Since we have likes, this will probably be the first metrics we'll look at
- I've also started working on indexing projects by libraries and which elements are used in it, so we'll be able to use that information soon

urish, 2021-01-14:
Another update:

There's a very initial version of a page that shows the top 10 liked projects:

https://wokwi.com/arduino/projects/___top_secret

The URL is temporary, of course, and will probably change soon. Once we have proper project thumbnails, perhaps I'll move it to the homepage

DiskDoctor, 2021-12-22:
I have not found any master list of projects or components?

urish, 2021-12-22:
Project - there's none at the moment, though it's on my radar

Components - The docs are currently the best reference: https://docs.wokwi.com

Some components are still missing, but they'll eventually be added there