wokwi/wokwi-features #41: not important: trying to check for errors on empty page pops error - can it be handled?

Puneethkumarsn, 2021-01-07:
open the quick start page:

https://wokwi.com/share

Click on the button: "check for error"

I understand it is right in its own way though t expect the setup function. 

[image]

[image]

urish, 2021-01-07:
How would you expect it to work in this case?

Puneethkumarsn, 2021-01-08:
if it is empty, check for errors button need not be active
OR
it can state, no content or file is empty

urish, 2021-01-08:
> if it is empty, check for errors button need not be active

Sounds good!

closed 2021-01-08 by urish