wokwi/wokwi-features #42: quick start link for wokwi arduino simulator project

Puneethkumarsn, 2021-01-07:
**Link**: https://wokwi.com/share

I feel that we should have a better page to start a project from scratch. something like an empty template. 
here, we can keep this page as is and give a button which will take to a new page where the user will also get the simulation window.

urish, 2021-01-07:
I'm thinking - maybe we can add several templates people can use?

urish, 2021-06-28:
Closing, as we already have it! 

* https://wokwi.com/arduino/new?template=arduino-uno
* https://wokwi.com/arduino/new?template=arduino-mega
* https://wokwi.com/arduino/new?template=arduino-nano
* https://wokwi.com/arduino/new?template=attiny85
* https://wokwi.com/arduino/new?template=pi-pico
* https://wokwi.com/arduino/new?template=micropython-pi-pico
* https://wokwi.com/arduino/new?template=franzininho
* https://wokwi.com/arduino/new?template=nano-rp2040-connect
*

closed 2021-06-28 by urish