wokwi/wokwi-features #43: text element option

Puneethkumarsn, 2021-01-08:
sometimes picture needs some text to save time

a label kind of feature which can be placed so that, 
1. I can tell how to give a long press
2. I can tell what the buttons are for
3. I can tell, what does the extreme ends of the potentiometer to
4. etc

stevesigma, 2021-01-22:
Maybe it would be used for global label assigned to any element, for example when sharing scheme.
Something like:
 <id="resistor1" type="wokwi-resistor" label="R1"...
 <id="btn1" type="wokwi-pressbutton" label="Change color"...

urish, 2021-01-23:
That's an interesting thought. So there are two alternatives:

1. Adding some kind of text/label primitive, which you can place in the diagram anywhere you like
2. Adding a new attribute to parts which will let you specify additional information. I'm not sure if I'd call it label, perhaps comment or description make more sense?

stevesigma, 2021-01-24:
1. would need fixed position, z-index, colors... that would make sometimes small problems. (when zooming?) Less code for You.
2. the position should be calculated on the fly while zooming. Labels could be connected with object by thin line/arrow. Style has to be autmatically. Should have some opaque background, could dissapear. More code for You.
name of attribute was just an example.

alto777, 2021-08-31:
I found this searching for a feature.

+1 on some kind of document annotation mechanism, not terribly important really, but once there are more than a few switches and LEDs it gets a bit of work to figure out which are which. Especially for others if you are sharing. 

alto777

rcwilcox, 2021-09-09:
> That's an interesting thought. So there are two alternatives:
>
> 1. Adding some kind of text/label primitive, which you can place in the diagram anywhere you like
> 2. Adding a new attribute to parts [...]

I like (1)

If attached, they would move if the part moved, which would be nice.  Also be nice to be able to to be attached (Relative Attribute) to a NET (e.g. Arduino Pin 11 - MOSI), should also be able to rotate and perhaps offset from attached object.  Visually connect to part as suggested by @stevesigma. Should also allow an absolute position (Fixed Attribute) test box for Diagram notes, usage etc.

matititam, 2021-09-13:
random thought - just wanted to pin it here

How about making the text note a mouse hover display (or when we click on the part (when the simulation is not running)

rcwilcox, 2021-09-14:
> random thought - just wanted to pin it here
> 
> How about making the text note a mouse hover display (or when we click on the part (when the simulation is not running)

That would help declutter (if you had lots of text labels), downside would be that any screen shots would not have the text visible.  My 'too much of anything is just enough' and 'retched excess' gene says - Have both with a **persistence** attribute of **static** or **dynamic**.    Of course, Fixed text objects (notes, tables, etc.) should be static.

M-square, 2021-09-25:
Of all the options above, I think a simple, independent (new) object that is just the text.

How it is handled in zoom is up to the implementation (scales or remains fixed size) whatever is simpler.

Text can have lots and lots of options (size, font, colour, alignment ...) - use the KISS principle.

marvinschwartz, 2022-07-12:
When you have lots of similar components on a screen you need some way to know what they are. eg. multiple stepper motors, multiple leds, multiple switches and buttons etc. This would make the diagrams far more useful and much better when sharing. Just a simple text element would be a significant improvement.

urish, 2022-08-26:
Update: we have a proof-of-concept implementation of a `wokwi-text` element:

https://wokwi.com/projects/341085054102930004

The details (element name, attrs, default behavior, etc) are subject to change - looking to get some feedback so we can ship this soon.

urish, 2022-09-21:
Apparently, the current implementation breaks the text differently depending on the screen sizes (e.g. desktop v.s. mobile), so we need some deterministic method of defining the size of the text element.

janoc, 2022-09-25:
Having the option 2 from https://github.com/wokwi/wokwi-features/issues/43#issuecomment-766184841 would be really useful if those labels could be actually attached to wires - **and exported** as part of the logic analyzer files too. It would make the analyzer a lot more useful when debugging multiple signals at once.

urish, 2022-09-25:
Thanks for the input @janoc, that's a good point! I'm curious - what have you been building/debugging with Wokwi?

janoc, 2022-09-25:
Trying some RP2040 PIO code. It seems Wokwi is the only thing around that has a sensible simulator available.

drf5n, 2023-03-03:
> Update: we have a proof-of-concept implementation of a `wokwi-text` element:
> 
> https://wokwi.com/projects/341085054102930004
> 
> The details (element name, attrs, default behavior, etc) are subject to change - looking to get some feedback so we can ship this soon.

For folks who ended up here while looking for documentation, the block of json in the diagram is:

    {
      "type": "wokwi-text",
      "id": "led1",
      "top": -40.7,
      "left": 67.67,
      "attrs": { "text": "This is an example of\nText Labels for Wokwi." }
    }

drf5n, 2023-03-16:
Does the element have extra attributes like:
```
"size": "10"   // canvas units? Font points?  
"justification": ""    // "", "left",  "center", "right" 
"font": ""                  // css type?
```

shaos, 2023-06-11:
Why attribute for proposed wokwi-text is called "text"? 
There is a perfect attribute that is doing similar thing for other parts - "label" 🙂
So technically object wokwi-text could be an empty part with a label 😉
From other side label is always 1-line text and wokwi-text could be multi-line (with \n separating lines) so this is a perfect time to make label multi-line to cover both use cases ;)

HerbieJay, 2024-11-23:
I would love to be able to place labels on wires int eh diagram window. Sometimes color coding is not enough and being to label the wiring network like in PCB design software would be helpful.

Jediweirdo, 2024-11-23:
I was not expecting the latest reply to be from an hour ago-- this is a lot more sought-after than I thought! The most nuclear way I could think of implementing text would probably be too much work to justify doing (see the bottom of this comment), so if I had to choose between an attribute and a component, I would side with team component hands down. You could base its connection on an "attachment" attribute that points to whatever object you connect it to, with the default connection being the background/canvas

Another thing I think would be cool would be the ability to add hyperlinks to comments that would "jump" the user to highlighted parts of the canvas or a highlighted line/section of code in a linked file.
rd
Using pictures in uploads would be cool too, but restrict HD picture uploads or the entire feature altogether to Wokwi subscribers because I bet it would bloat server costs like crazy. 

--- _Nuclear comment idea_:

Add a comment key value to every part's `chip.json`. It would have 5 items:

-  Attachment
Attachment is how you would decide the comment's point of origin when placing it on the canvas. It accepts 2 types of values:
    1. Pin Names
        - Uses the pin name to find the pin's location on the chip's SVG and uses it as the comment's point of origin
    2. The Chip's Name
        - Uses the exact center of the board as the comment's point of origin
Attaching comments to nothing defaults their attachment to 2nd option. Attaching to the canvas could be done through a third option, but it would get kinda messy. So instead, give the canvas a `.chip.json`! I'm sure there's things about the canvas people would like to customize anyway, like the canvas's color, grid unit measure (like using inches instead of milimeters), gridlock size, etc.
- Description
Where you write the text for your comment. Doing rich text formatting similar to something like an MD file would be pretty cool, but there's a couple of proprietary syntaxes that would be cool to impliment, like...
    * `[code jump text]{line_highight_start-line_highlight_end}`,  `[canvas jump text]{top_pos-left_pos}`, and `[canvas jump text]{name_for_chip_to_jump_to_and_highlight}`
    * `[pin highlight text here]{->Pin_name}`

urish, 2025-01-14:
wokwi-text has now been officially released, you can select it from the drop down menu:

[image]

and when you click on a text element, there's a dialog where you can change the text.

closed 2025-01-14 by urish

drf5n, 2025-01-25:
Cool!  I noticed it has a entry on the table:

[image]

and that the text is editable and rotatable if you click on it:

[image]

Thanks