wokwi/wokwi-features #44: Precompiler syntax error displays error after error correction. Code runs though.

stevesigma, 2021-01-08:
I am human = I make mistakes. (I am superHuman)
1. I make mistake (missing semicolon at the end)
2. I click Green arrow Start simulation 🟢
3. Popup window "Build failed" and marked first error ❗ "expected ',' or ';' before 'static'" with red lines
4. I correct missing semicolon, click "❌" to clear error.
5. I save project (it does not matter)
5. I click Green arrow Start simulation 🟢
6. It will compile and simulation starts, but at the place of previous error is raised slightly different ❗ "expected initializer before 'extern'"
There is no error next. Maybe some cache is not refreshed or compiling file is not changed.

urish, 2021-01-08:
Reproduced; now trying to figure out why :-)

urish, 2021-01-08:
I believe I found the cause and fixed it, so now if you close the error window, and recompile without any errors it won't pop again.

closed 2021-01-08 by urish