wokwi/wokwi-features #46: feedbaxck from tweaking4all.com webmaster

Puneethkumarsn, 2021-01-11:
Context:
link: https://wokwi.com/arduino/projects/287311734683730445

Two suggestions to an otherwise seemingly working well simulation:

1) The LEDs are displayed as rectangle blocks, maybe it would be abetter idea to make them round?

When running this sketch on a real Arduino+LED strip, **one would see the individual LEDs, whereas with your simulation we see one solid line.**
It would look better if one could see the individual LEDs in the simulation as well. 

2) **The effects sometimes flicker, where this doesn’t happen in a real setup.**

Naturally, this could be a browser issue or a refresh issue.

urish, 2021-01-11:
Thanks for the feedback! 

Basically, working on a new `wokwi-led-matrix`, which should solve the first issue (maybe also the second?):

https://wokwi.com/arduino/projects/287433189177164300

Puneethkumarsn, 2021-05-17:
I think the issue is addressed. Voting to close

closed 2021-05-17 by urish