wokwi/wokwi-features #47: the big number in the weblink is not readable

Puneethkumarsn, 2021-01-11:
Hi

Many people have shown interest in making the weblink readable? is it do-able?

instead of a number, like in- https://wokwi.com/arduino/projects/287311734683730445 
can it be: https://wokwi.com/arduino/projects/all-simulations-in-one-xxxxxxx 
it is a good idea? can the user specify part of the file name?

Puneethkumarsn, 2021-05-17:
voting for closure. The .ino file name is okay.

urish, 2021-05-17:
Thanks! 

I'm thinking of changing the project link in the feature, to https://wokwi.com/projects/XXXX - removing the /arduino, since we have Pi Pico now and probably MicroPython as well at some point. So perhaps we'll also change the format of the XXXX when we change the links.

closed 2021-05-17 by urish