wokwi/wokwi-features #52: Simulations unreliable in Safari on Mac OS X

costyn, 2021-01-15:
These don't work, for example:
* https://wokwi.com/arduino/libraries/FastLED/DigitalRain
* https://wokwi.com/arduino/libraries/FastLED/Fire2012
* https://wokwi.com/playground/ssd1306

urish, 2021-01-15:
Possibly related: wokwi/wokwi-elements#50

urish, 2021-01-16:
SSD1306 issue fixed! 

As for the FastLED examples, they work fine for me under Safari 13.0. Can you describe what issues they have on your computer?

costyn, 2021-01-17:
I don't use Safari myself (so not problem for me), but wanted to demo something to a friend and they were using Safari and it was glitching. What happens in Safari is the CPU measurement spikes to 150% in 2 seconds and the animation stops, also the timer crawls to a halt. Maybe there's a throttling function in Safari? Making sure that a webpage doesn't start hogging resources?

urish, 2021-01-17:
Yes, I don't use Safari either, and unfortunately can't reproduce the CPU/timer issue when testing here.

Which version of Safari did you test it with?

costyn, 2021-02-03:
Sorry I totally forgot to reply to this. Version 14.0.2 (16610.3.7.1.9)  (what comes with Mac OS X Big Sur)

urish, 2021-02-05:
Can you please test with this version and see if you observe any difference?

https://wokwi-new-git-workers.wokwi.vercel.app/arduino/libraries/FastLED/Fire2012

costyn, 2021-02-05:
This works fine on my Safari now! Smooth and doesn't stop. You fixed it!