wokwi/wokwi-features #53: diagram.json overwritten with content from another editor tab

sutaburosu, 2021-01-15:
I've seen this twice now.  The first time, 2-3 days ago, I was copying and pasting code, so I assumed I made a mistake.  The Undo buffer was empty, which I thought was odd.

It just happened again.  I opened a browser, went to `/arduino/projects` and opened a project.  I made a 1 character edit to diagram.json.  Then I dragged a link from Discord to open another browser tab.  Returning to the first tab I find diagram.json now has the same content as the .ino,  and the Undo buffer is empty.

Following those steps again doesn't reproduce it for me.  I forgot to look in the JS console before restarting the browser.  I won't forget if it happens again.

urish, 2021-01-21:
I believe I managed to reproduce this - it happens to me whenever I switch to `diagram.json` tab before the editor is fully loaded. I'll keep investiaging

urish, 2021-01-23:
Fixed :-)

closed 2021-01-23 by urish