wokwi/wokwi-features #54: serial plotter crashes with non-numeric input

sutaburosu, 2021-01-16:
Run https://wokwi.com/arduino/libraries/DHT-sensor-library/DHT-Tester and enable the serial plotter.

Extracting and plotting the numbers from any string would be great.  Of course, it's impossible to do the right thing in all cases,  but a [simple solution](https://xkcd.com/208/) would work nicely for this sketch and many others.

Here are the pieces that fell out:
```
[Serial] Humidity: 40.00%  Temperature: 24.00Â°C 75.20Â°F  Heat index: 23.50Â°C 74.30Â°F
framework.9707fddd9ae5927c17c3.js:1 TypeError: Cannot read property 'data' of undefined
    at e (a6f8e7f9e9fa63ed9c17d358cd68c322520f2703.e64fcac466db7b7b0074.js:1)
    at a6f8e7f9e9fa63ed9c17d358cd68c322520f2703.e64fcac466db7b7b0074.js:1
    at Oi (framework.9707fddd9ae5927c17c3.js:1)
    at t.unstable_runWithPriority (framework.9707fddd9ae5927c17c3.js:1)
    at jl (framework.9707fddd9ae5927c17c3.js:1)
    at Ti (framework.9707fddd9ae5927c17c3.js:1)
    at Ie (framework.9707fddd9ae5927c17c3.js:1)
    at Xt (framework.9707fddd9ae5927c17c3.js:1)
ou @ framework.9707fddd9ae5927c17c3.js:1
a.componentDidCatch.n.callback @ framework.9707fddd9ae5927c17c3.js:1
ca @ framework.9707fddd9ae5927c17c3.js:1
pu @ framework.9707fddd9ae5927c17c3.js:1
Pi @ framework.9707fddd9ae5927c17c3.js:1
t.unstable_runWithPriority @ framework.9707fddd9ae5927c17c3.js:1
jl @ framework.9707fddd9ae5927c17c3.js:1
Ci @ framework.9707fddd9ae5927c17c3.js:1
pi @ framework.9707fddd9ae5927c17c3.js:1
(anonymous) @ framework.9707fddd9ae5927c17c3.js:1
t.unstable_runWithPriority @ framework.9707fddd9ae5927c17c3.js:1
jl @ framework.9707fddd9ae5927c17c3.js:1
Ql @ framework.9707fddd9ae5927c17c3.js:1
$l @ framework.9707fddd9ae5927c17c3.js:1
Oi @ framework.9707fddd9ae5927c17c3.js:1
t.unstable_runWithPriority @ framework.9707fddd9ae5927c17c3.js:1
jl @ framework.9707fddd9ae5927c17c3.js:1
Ti @ framework.9707fddd9ae5927c17c3.js:1
Ie @ framework.9707fddd9ae5927c17c3.js:1
Xt @ framework.9707fddd9ae5927c17c3.js:1
main-6a415dde87ce0002a976.js:1 TypeError: Cannot read property 'data' of undefined
    at e (a6f8e7f9e9fa63ed9c17d358cd68c322520f2703.e64fcac466db7b7b0074.js:1)
    at a6f8e7f9e9fa63ed9c17d358cd68c322520f2703.e64fcac466db7b7b0074.js:1
    at Oi (framework.9707fddd9ae5927c17c3.js:1)
    at t.unstable_runWithPriority (framework.9707fddd9ae5927c17c3.js:1)
    at jl (framework.9707fddd9ae5927c17c3.js:1)
    at Ti (framework.9707fddd9ae5927c17c3.js:1)
    at Ie (framework.9707fddd9ae5927c17c3.js:1)
    at Xt (framework.9707fddd9ae5927c17c3.js:1)
ue @ main-6a415dde87ce0002a976.js:1
fn @ main-6a415dde87ce0002a976.js:1
value @ main-6a415dde87ce0002a976.js:1
a.componentDidCatch.n.callback @ framework.9707fddd9ae5927c17c3.js:1
ca @ framework.9707fddd9ae5927c17c3.js:1
pu @ framework.9707fddd9ae5927c17c3.js:1
Pi @ framework.9707fddd9ae5927c17c3.js:1
t.unstable_runWithPriority @ framework.9707fddd9ae5927c17c3.js:1
jl @ framework.9707fddd9ae5927c17c3.js:1
Ci @ framework.9707fddd9ae5927c17c3.js:1
pi @ framework.9707fddd9ae5927c17c3.js:1
(anonymous) @ framework.9707fddd9ae5927c17c3.js:1
t.unstable_runWithPriority @ framework.9707fddd9ae5927c17c3.js:1
jl @ framework.9707fddd9ae5927c17c3.js:1
Ql @ framework.9707fddd9ae5927c17c3.js:1
$l @ framework.9707fddd9ae5927c17c3.js:1
Oi @ framework.9707fddd9ae5927c17c3.js:1
t.unstable_runWithPriority @ framework.9707fddd9ae5927c17c3.js:1
jl @ framework.9707fddd9ae5927c17c3.js:1
Ti @ framework.9707fddd9ae5927c17c3.js:1
Ie @ framework.9707fddd9ae5927c17c3.js:1
Xt @ framework.9707fddd9ae5927c17c3.js:1
```

urish, 2021-01-16:
Reproduced

urish, 2021-01-16:
Fixed! Thanks :-)

closed 2021-01-16 by urish