wokwi/wokwi-features #55: Sensors for arduino - Microphone

Puneethkumarsn, 2021-01-16:
do we already have a microphone as the simulated element? 

can we use the PC/mobile microphone to emulate real microphone on wokwi?

I would like to try adding the mic to fastLED projects!

urish, 2021-01-16:
That would be a cool idea!

One challenge is that the simulation does not always run at 100%, so if we stream real-time microphone data, it might get distorted (frequency skew). But maybe it's still better than nothing?

TomasHubelbauer, 2021-05-01:
Could also be cool to allow selecting a pre-recorded audio file or when placing the microphone, show a UI for recording a sample to store for later replay, so that the read from the microphone/sample playback could happen in sync with the simulation.

Puneethkumarsn, 2021-05-02:
@TomasHubelbauer That is indeed a good idea..

atuline, 2021-05-09:
Even if you can't run FFT, let's say due to distortion, you can get a LOT done with volume and peak detection.

urish, 2021-05-11:
First prototype: https://wokwi.com/arduino/projects/298333297784652296

There's still no user interface element, but it should capture the audio and turn the LED on when there are peaks (also plot the waveform to the Serial Plotter)