wokwi/wokwi-features #59: wokwi0led-matrix element to be added to a list

Puneethkumarsn, 2021-02-09:
[image]

Puneethkumarsn, 2021-05-17:
I think we can quickly add this to the list and close
[image]

urish, 2021-05-18:
I'm not sure if we'll be keeping the led-matrix. We already have neopixel-canvas that does pretty much the same.

We need to come up with good naming scheme and stick to it. Right now we have:

* wokwi-neopixel
* wokwi-neopixel-matrix
* wokwi-neopixel-canvas
* wokwi-neopixel-strip
* wokwi-led-matrix
* wokwi-led-ring

All of them work with WS2812B (NeoPixel) protocol. I find it confusing.

drf5n, 2022-04-16:
> Right now we have:
> 
> * wokwi-neopixel
> * wokwi-neopixel-matrix
> * wokwi-neopixel-canvas
> * wokwi-neopixel-strip
> * wokwi-led-matrix
> * wokwi-led-ring

Only  wokwi-neopixel, wokwi-led-ring and wokwi-led-matrix are on the UI's + list:

[image]

And none of them are listed on the parts sidebar at https://docs.wokwi.com/parts/wokwi-led

[image]

urish, 2026-02-14:
Done.