wokwi/wokwi-features #61: options in the editor tab similar to Arduino IDE

Puneethkumarsn, 2021-02-15:
[image]

* Comment/uncomment
* increase and decrease indent
something to match with the Arduino editor experience

sutaburosu, 2021-02-15:
I'm not sure matching the Arduino IDE is desirable. :)

The Monaco editor used here is the same editor used in VS Code. Press F1 to call up the command palette, which is useful index to keyboard shortcuts.

To comment/uncomment, use `Ctrl /`.

For indentation, use `Ctrl ]` and `Ctrl [`.

Puneethkumarsn, 2021-05-17:
I think this issue must be deleted. The Editor has more than enough feature for everyone to work with already

closed 2021-05-17 by urish