wokwi/wokwi-features #64: Option to upload the ino, .h and .cpp files

Puneethkumarsn, 2021-03-13:
The **request** is to have an option to upload the files to the project.

Use case: For a multi-file project, there will be multiple files to be created manually 

- each file name has to be entered manually
- and the contents of the file has to be copied to the editor

The **benefit** of this feature
it will help the user to easily upload the files, especially in multifile projects

A **bonus** feature will be to upload all the project files at once :)

urish, 2021-03-13:
Thanks for the suggestion! 

How would you like to upload the files?

Puneethkumarsn, 2021-03-13:
an option in the menu itself
[image]

Puneethkumarsn, 2021-05-18:
another feature along with the upload file is to mention the link of the file. User shall be able to give the file link from GitHub or another site for example. as long the type of the file is supported.

urish, 2021-09-30:
Thanks! This is now implemented:

[image]

closed 2021-09-30 by urish