wokwi/wokwi-features #65: Add option to serial monitor to change new line behavior

ooomens, 2021-03-15:
It would be really nice if the user could decide, which control char were submitted at the end a user input.
The Arduino IDE and TinkerCAD only submit "\n", which I think is more intuitive than submitting both CR and NL. So maybe also make this common and set the "Only NL" option as default.

Cheers

urish, 2021-03-18:
Thanks for the suggestion! 

A prototype of this feature is now available on the following branch:

https://serial-options.preview.wokwi.com

It enables you to configure your serial monitor by adding a `"serialMonitor"` object to your diagram.json, e.g.:

```json
  "serialMonitor": { 
    "newline": "lf"
  }
```

the possible values are:
- `none` - don't add any newline characters
- `cr` - add Carriage Return ("\r")
- `lf` - add Linefeed ("\n")
- `crlf` - add both ("\r\n")

As for changing the default - I want to confirm first that it's "\n" even on Arduino for windows. If it is so, we'll change it as you suggested.

urish, 2021-03-20:
Default behavior changed to "\n".

urish, 2021-03-20:
Merged + [docs updated](http://docs.wokwi.com/guides/serial-monitor#newline)

closed 2021-03-20 by urish