wokwi/wokwi-features #66: Add option to open serial monitor without printing into it first

ooomens, 2021-03-15:
If you write a program based on user input it would be helpful to have a tick box or just a toggle button to open the serial monitor so you don't have to print something first to have the window open.

urish, 2021-03-18:
A prototype of this feature is now available on the following branch:

https://serial-options.preview.wokwi.com

It enables you to configure your serial monitor by adding a `"serialMonitor"` object to your diagram.json, e.g.:

```json
 "serialMonitor": { 
    "display": "always"
  }
```

the possible values are:
- `auto` - the default behavior, show the serial monitor once some input has been received
- `always` - always show the serial monitor
- `never` - never show the serial monitor
- `plotter` - always show the serial plotter

urish, 2021-03-20:
This has just been merged! Keeping it open until we update the docs

urish, 2021-03-20:
[Docs updated too](http://docs.wokwi.com/guides/serial-monitor#display)

closed 2021-03-20 by urish