wokwi/wokwi-features #71: Linker errors only shown on Console

M-square, 2021-03-27:
Sometimes you get a Build Failed messagebox, but nothing is indicated in the code panel. Looking at the console window the text shows the failure is found by the linker or assembler. Typically a label cannot resolve
Example assembler .S code and error found by this phase:  (the .ino file is empty)
```
.global main
main:
  nop
  brne 1f
  nop
```
```
tmp/ccy3otqu.s: Assembler messages:
/tmp/ccy3otqu.s: Error: local label `"1" (instance number 12 of a fb label)' is not defined
Error during build: exit status 1

```

urish, 2021-03-29:
Related: #67

urish, 2021-03-30:
Thanks for reporting! They'll now also appear in the code editor. We can't really associate them with a specific line / file, so they'll appear on the first line of the main sketch file

closed 2021-03-30 by urish

M-square, 2021-04-03:
Tested and verified.
I missed the bit about it being in the sketch file, i.e. the first .INO, but the tab turns red, so it is just a question of being observant.

(NB: Now I have an issue with github, I didnt get notification of your status change, haven't checked why yet)