wokwi/wokwi-features #72: Wiring bug on Safari

tawjaw, 2021-03-31:
Hello, 
The wiring on Safari has bugs. 
The following is the led example. Left on safari, right on Chrome.  

[image]

If I inspect the SVG element for the wiring part I get different SVGs from safari vs chrome 

[image]

urish, 2021-04-25:
I think I may fixed it in the new diagram editor branch: 

https://diagram-editor.preview.wokwi.com/arduino/libraries/demos/blink-mega

I couldn't reproduce the original issue, but I'd love to know if the new branch fixes this issue for you?

tawjaw, 2021-04-26:
The link you shared loads for a while and then gives me "An unexpected error has occurred." on safari and chrome. 

This works well though. 
https://diagram-editor.preview.wokwi.com/arduino/libraries/demo/blink

urish, 2021-04-26:
My bad, I pasted the wrong link. The correct one is: https://diagram-editor.preview.wokwi.com/arduino/libraries/demo/blink-mega

tawjaw, 2021-04-26:
the wiring is working fine. it's fixed

urish, 2021-04-26:
Hooray! Thank you :)

closed 2021-04-26 by urish