wokwi/wokwi-features #75: Github integration

patrickleboutillier, 2021-04-16:
It would great if we could point Wokwi to a github repo to load the code from there, and then commit the changes.

Developping on Wokwi is really faster than on a real device, but copying the code back and forth between Wokwi and the Arduino IDE to try in out on a real device can be a pain, especially for largeer projects with multiple files.

If Wokwi could read/write code to a GH repo it would be really helpful.

urish, 2022-01-14:
Good point, thanks for bringing this up @djedu28 !

leftCoast, 2022-01-14:
Doesn't the "upload library" stuff you just got working, kinda' make this a moot point? I can basically upload my entire development environment now. And that's a ton of files.

jflamy, 2022-02-13:
I work the other way round.  I download project as zip and throw all the unzipped files on top of the existing git-managed ones.  Git spots which ones have actually changed and my IDE shows them.  I can compare with HEAD, etc., and then commit.

urish, 2022-02-13:
That's actually pretty clever, @jflamy. Thanks for sharing!

urish, 2022-10-24:
The [Visual Studio Code integration](https://wokwi.com/features#feature-1368676003) has been getting a lot traction lately, and it will also enable a straightforward workflow with Git/GitHub.

If anyone is interested in moving over their votes from this issue to the VSCode integration request, please let me know and I'll assist.

jflamy, 2022-10-24:
Please spend all my available votes on this feature. As if I had been
voting for the last several months. I will be back soon !

urish, 2022-10-25:
> Please spend all my available votes on this feature. As if I had been voting for the last several months. 

Which one, the VSCode or the GitHub one? Is there anything that stops you from going to wokwi.com/features and voting there?

Twistx77, 2022-11-24:
> The [Visual Studio Code integration](https://wokwi.com/features#feature-1368676003) has been getting a lot traction lately, and it will also enable a straightforward workflow with Git/GitHub.
>
> If [...]

Agree, using the VSCode integration you don't need anymore Github integration on the online editor, you could just use Git directly from your PC or VSCode or your preferred Git Client and you don't need to even upload your code to the website.

apollolabsdev, 2024-07-19:
Has Github integration for the website been abandoned with VSCode as the alternative? A use case still exists where syncing the website with GitHub is beneficial. In my case, with educational material. I have to maintain two separate streams for web-based material distributed to students vs. git synced with VS Code. 

If a method can somehow be introduced for the web-based interface to sync or retrieve the latest code off a repo. Sort of like a one-way sync. Without necessarily having full (or any) git features.

urish, 2024-07-28:
Yes, we're focusing now on VS Code (and similarly, GitHub Codespaces / GitPod). 

We're also working on a way to simulate a given binary firmware directly from a URL, e.g.: https://wokwi.com/experimental/viewer?diagram=https://gist.githubusercontent.com/urish/c3d58ddaa0817465605ecad5dc171396/raw/ab1abfa902835a9503d412d55a97ee2b7e0a6b96/diagram.json&firmware=https://github.com/georgik/esp32-graphical-bootloader/releases/latest/download/graphical-bootloader-esp32-s3-box.uf2

apollolabsdev, 2024-08-04:
Awesome! Could that potentially be expanded to update the web-based URL project files? Maybe even through API.