wokwi/wokwi-features #76: RAM usage - realtime

Puneethkumarsn, 2021-04-17:
a new feature request to monitor RAM usage.

* Projects sometimes crash without reason (sometimes the reason will out of memory) 
* A reset will make it work, but the issue will be hard to find
* basic information such s maximum RAM used until now, will give an idea to the user

urish, 2021-06-25:
@sutaburoso I remember you had some idea how to implement it in the past. Or am I wrong?

sutaburosu, 2021-06-26:
Everything I learned and more is in the [MemoryFree library](https://github.com/McNeight/MemoryFree/blob/master/MemoryFree.cpp).  It uses the address of the local `free_memory` variable as a surrogate for the Stack Pointer.  It also walks malloc's internal structures to account for the free spaces within the heap, but I wouldn't expect the sim to do that. Free fragments within the heap are no use when the stack is colliding with heap's arena. :)

In essence, all that's needed is the stack pointer and the addresses of the symbols `__heap_start` and `__brkval`. The logic will almost identical to that in freeMemory():
```cpp
  if (__brkval == 0)
    free_memory = SP - &__heap_start;
  else
    free_memory = SP - __brkval;
```

I believe the difficulty lies in doing this without crippling performance.  Perhaps cache the lowest value of SP since the last check,  and only infrequently compare it to __brkval?  Perhaps checking only on `RET` would be useful enough.

One other tiny snag: if `malloc()` or `new` are never used in the sketch, the symbol `__brkval` won't be in the ELF.

urish, 2021-06-27:
Thanks Steve! 

I'm thinking on another direction: maybe if we keep a set of all the memory location ever written to, and we see that this set contains more than 85% of the address space, then we can warn the user there might be a memory issue?

I can see two potential problems here though:
1. What if some memory is allocated but never initialized, so we don't see any writes to those locations? I'm not sure this can actually happen, so this is just an idea.
2. It will probably have false negative if things just move around the memory during the lifetime of the program, as it'll touch many locations but they will later be free again.

Not sure which approach works better - reading SP is easy, and `__brkval` / `__heap_start` could be a bit more tricky (we'll probably need to fetch to ELF file for that, which can increase the time it takes for a project to start running).

urish, 2022-03-26:
Closing as this is technically challenging to implement. As a workaround, users who suspect they're running into memory issues can run their project with larger SRAM amount (using `attrs: { "__fakeRamSize": "32768" }`) and observe if the project behaves differently.

closed 2022-03-26 by urish