wokwi/wokwi-features #77: Undo feature for simulator window

Puneethkumarsn, 2021-04-24:
I would love to undo the movement of the components I do in the simulator window.

sutaburosu, 2021-08-20:
I'm not sure if work has already been done on this, but my experience is that Wokwi *can* undo/redo dragging components in the diagram. Just click in the `diagram.json` text tab before pressing ctrl-z or ctrl-shift-z.

Mixing manual edits to `diagram.json` and dragging in the GUI editor seems to work fine too. Starting/stopping the sim presents no problems either.

The only problem I've noticed comes when any other text tab is edited. `diagram.json` seems to lose its undo buffer at that moment.

Adding the ctrl-z and ctrl-shift-z shortcuts to the diagram would be great, if they could somehow affect only the `diagram.json` Monaco tab. If this is unreasonably difficult to achieve, even a popup directing the user to do it themselves may be useful.

urish, 2022-10-03:
Undo is ready! Keeping this issue open until we get to update the docs

matititam, 2022-10-06:
looks good. tried with a few projects