wokwi/wokwi-features #78: Can it import a firmware?

yiivon, 2021-04-24:
That would be more wonderful if wokwi-playgrounds can importing a firmware and run it directly, it possibles?

urish, 2021-05-10:
Hi Yiivon,

It's now possible to import a HEX file:

1. Click on the code editor and press "F1"
2. Select "Load HEX File and Start Simulation..."
3. Choose a HEX file to load

[image]

Would love to hear your feedback about this new feature!

guymcswain, 2021-11-29:
Is this feature still available - "Load HEX File and Start Simulation..."
I'm not seeing it in the code editor on my browser (MS Edge nor Google Chrome).

urish, 2021-11-29:
The feature is indeed still available. What do you see when you press F1 inside the code editor?

guymcswain, 2021-11-29:
I get a drop down list: "Add cursor above", "add cursor below", etc, but not this particular item.  I'm a noob with wokwi.

urish, 2021-11-29:
That's ok, it's a good start. Then, what happens when you type the word "HEX" inside this drop down? what options do you see then?

guymcswain, 2021-11-29:
That is it.  I see it now.  Thank you!

urish, 2021-11-29:
You are welcome :)

green-mcu, 2023-09-06:
**urish**, is it possible to store HEX on the server?

urish, 2023-09-06:
On Wokwi's server? What's your use case for that?

green-mcu, 2023-09-06:
Yes, on Wokwi's server. So that anyone can do HEX without a sketch.

green-mcu, 2023-09-06:
So that everyone can run the program without having the source code.

urish, 2023-09-06:
Actually, we do want to encourage people to share their source code.

An alternative for people who want to share the hex without the source code is to put the files on GitHub, where they can be viewed using Wokwi for VSCode. Here is an example for such setup: https://github.com/wokwi/esp-idf-hello-world

hayschan, 2023-09-18:
> So that everyone can run the program without having the source code.

This is also a use case for me. I would like to do training for my end users without giving them the source code nor the `.hex` file. Wokwi will be a brilliant platform for that.

> An alternative for people who want to share the hex without the source code is to put the files on GitHub, where they can be viewed using Wokwi for VSCode.

Using the browser is always the first option for non-technical users. Will be great if that is achievable.

urish, 2023-10-02:
> This is also a use case for me. I would like to do training for my end users without giving them the source code nor the .hex file. Wokwi will be a brilliant platform for that.

Thanks for explaining your use case! If we launched that a paid feature, would that work for you?

hayschan, 2023-10-05:
> If we launched that as a paid feature, would that work for you?

Yep, I have no problem for paying. I understand advanced feature like these deserve being in the subscription plan.