wokwi/wokwi-features #89: I am unable to edit code on Safari macOS High Sierra

E-Owsteen, 2021-05-07:
So I am on a Macbook air 2017 which is 4~ years old but still working great. The mac uses macOS High Sierra version 10.13.6. I'm using safari version 13.1.2 and I am unable to edit the code on the ".ino" and ".json" files. When I open any projects up, I am able to add parts using the purple plus and move them around using my mouse but when it comes to editing the text, it doesn't work. I don't see the text cursor and any input into my keyboard doesn't register into text. I can however highlight the code and run the simulation. 

[image]

urish, 2021-05-07:
Thanks for reporting!

Seems like the editor is unable to load for some reason. Can you open the Safari dev tools (Option + ⌘ + C) and check if there's some error message or warning that appears while the page is loading?

sutaburosu, 2021-05-08:
The editor not loading reproduces for me here in the latest Safari (13.1.2) on macOS High Sierra.  Safari 14.x doesn't have this problem, but I think that needs a more recent macOS.

I spent 5-minutes trying to find how to press the dev-tools key combo in virtmanager, or find a way to trigger the dev console via menus.  I even tried thinking differently; it didn't help.

[This Monaco issue](https://github.com/microsoft/monaco-editor/issues/2461) seems to describe exactly the same symptoms. It has a screenshot of the dev console. The error seems to indicate this a problem that only the Monaco and/or Safari devs can reasonably fix.

urish, 2021-05-08:
Thank you so much for investigating it, @sutaburosu! 

I deployed a workaround, where it checks for this issue, and then goes back to a previous version of the editor known to work (0.21.3, according to the issue you linked):

```javascript
function isOldSafari() {
  try {
    new RegExp('([^\\b_])(\\p{Lu})(\\p{Ll})', 'gmu');
    return false;
  } catch (err) {
    return true;
  }
}
```

I hope this fixes the issue for now!

sutaburosu, 2021-05-08:
Yes, this works for me in Safari 13 and 11.  Safari's dev console visibility is controller by Preferences -> Advanced -> Show debug.  :+1:

urish, 2021-05-08:
Hooray! 🎉

closed 2021-05-08 by urish