wokwi/wokwi-features #90: IR Receiver / Transmitter / Remote Simulations ?

jimf99, 2021-05-09:
Would be nice to have an IR Receiver, Transmitter, Remote added.
Thank You!
Kind Regards,
JimFyyc Calgary Alberta Canada

urish, 2021-05-16:
Thanks for suggesting! As I first step, I added the IRremote library to the simulation. 

What IR remote model should we add?

jimf99, 2021-05-16:
1) NEC IR Remote Is popular and easy to understand the IR Codes and timings.
Once you have NEC then SAMSUNG Remotes are popular too.

Hope this helps.

urish, 2021-05-16:
Thanks! Do you have a link to a drawing or photo?

jimf99, 2021-05-16:
https://cdn-shop.adafruit.com/970x728/389-03.jpg

jimf99, 2021-05-16:
Generic IR Remote

https://cdn-shop.adafruit.com/970x728/389-03.jpg

urish, 2021-05-18:
Good news: IR Receiver + Remote control are ready!

* Quick demo: https://wokwi.com/arduino/projects/298922033226449421
* [IR Receiver Docs](https://docs.wokwi.com/parts/wokwi-ir-receiver)
* [IR Remote Docs](https://docs.wokwi.com/parts/wokwi-ir-remote)

closed 2021-05-18 by urish