wokwi/wokwi-features #92: feature request - Search bar on home apge to search

Puneethkumarsn, 2021-05-14:
Even I don't have a 100% need for it, but I wanted to quickly search on the Wokwi site, whether we have any projects involving servo motors. 
Shall we have a search button?

urish, 2022-03-26:
Duplicate of #73

Fa5g, 2023-11-10:
It would be a good idea to have a button through which we can search the community of users for specific projects according to our needs and thus learn from what others have developed.

drf5n, 2024-01-16:
How does one vote for this?  

I would really, really like a method to search for projects, either globally, or within my own set of projects.  Fiding even my own projects with the flat 2-column graphic report under https://wokwi.com/dashboard/projects is very slow and awkward to use.