wokwi/wokwi-features #93: Regression: pinch-to-zoom zooms the sim, but also zooms the browser

sutaburosu, 2021-05-18:
To reproduce, visit any sim on a mobile device and pinch-zoom on the right-hand "sim" pane.  This affects both Chrome and FF.

I believe the previous behaviour was to swallow the pinch-zoom event, so only the sim and not the browser reacted to it.

Puneethkumarsn, 2021-05-19:
In my PC with Firefox, the two pinch-zoom zooms in and out the whole browser. 
but, in the rightmost of the mousepad, if I use two-finger move up or move down, the simulator window zooms as expected

urish, 2021-06-09:
I've applied a CSS rule to prevent the browser for intercepting the zoom event. Does the issue still exist?

matititam, 2021-06-10:
I see the issue present still on Firefox.
The issue only when I use the mousepad on my laptop. With the actual mouse, zoom happens to work fine.

[image]
This works. when I move two fingers up or down, it zooms in or zooms out only the simulation window

[image]

This zooms in or out the whole browser

urish, 2022-01-27:
Is this still relevant?

urish, 2022-03-27:
Closing due to lack of activity

closed 2022-03-27 by urish