wokwi/wokwi-features #94: The sim diagram's default zoom level could be capped at some sensible maximum

sutaburosu, 2021-05-18:
When starting from [a new sketch](https://wokwi.com/arduino/new), on some screens the new MCU is huge and dominates the diagram tab. 
[image]

Perhaps the default zoom level could be capped at some level,  so the size of the elements is never drastically bigger than in real life.

urish, 2021-05-18:
Thanks for reporting!

Limited the default zoom level to x1.5. 

Is it still too big? Or perhaps too small now?

sutaburosu, 2021-05-18:
This seems much better from my perspective. I didn't realise that I wanted this until it was suggested.

@Puneethkumarsn how is it for you?

Puneethkumarsn, 2021-05-19:
It is awesome here :)

sutaburosu, 2021-05-19:
Thanks for the feedback, and thanks, Uri, for the quick turnaround.

closed 2021-05-19 by sutaburosu