wokwi/wokwi-features #96: LiquidMenu and config file

fred-76, 2021-05-24:
Hello

There is a config file in the LiquidMenu library and it makes errors...

Cheers

Fred

urish, 2021-05-24:
Hi Fred,

Can you please share an example?

fred-76, 2021-05-24:
```
//Compatible with the Arduino IDE 1.0
//Library version:1.1
#include <LiquidCrystal_I2C.h>
#include <LiquidMenu.h>

LiquidCrystal_I2C lcd(0x27,20,4);  // set the LCD address to 0x27 for a 16 chars and 2 line display

void setup()
{
  lcd.init();                      // initialize the lcd
  // Print a message to the LCD.
  lcd.backlight();
  lcd.setCursor(0,0);
  lcd.print("Hello World!");
}

void loop()
{
}
```

Ends up with error:

```
In file included from /home/hexi/Arduino/libraries/LiquidMenu/src/LiquidScreen.cpp:30:0:
/home/hexi/Arduino/libraries/LiquidMenu/src/LiquidMenu.h:54:123: note: #pragma message: LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.
 # pragma message ("LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.")
                                                                                                                           ^
In file included from /home/hexi/Arduino/libraries/LiquidMenu/src/LiquidSystem.cpp:30:0:
/home/hexi/Arduino/libraries/LiquidMenu/src/LiquidMenu.h:54:123: note: #pragma message: LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.
 # pragma message ("LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.")
                                                                                                                           ^
In file included from /home/hexi/Arduino/libraries/LiquidMenu/src/recognizeType.cpp:7:0:
/home/hexi/Arduino/libraries/LiquidMenu/src/LiquidMenu.h:54:123: note: #pragma message: LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.
 # pragma message ("LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.")
```

urish, 2021-05-24:
Oh, these are not errors. They are just notices printed by the code, and the simulation should run regardless of these messages. They also appear when you compile the same sketch via the Arduino IDE:

```
In file included from C:\Users\Uri\AppData\Local\Temp\arduino_modified_sketch_620015\sketch_may24a.ino:4:0:
C:\Users\Uri\Documents\Arduino\libraries\LiquidMenu\src/LiquidMenu.h:54:123: note: #pragma message: LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.
 # pragma message ("LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.")
                                                                                                                           ^
In file included from C:\Users\Uri\Documents\Arduino\libraries\LiquidMenu\src\recognizeType.cpp:7:0:
C:\Users\Uri\Documents\Arduino\libraries\LiquidMenu\src\LiquidMenu.h:54:123: note: #pragma message: LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.
 # pragma message ("LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.")
                                                                                                                           ^
In file included from C:\Users\Uri\Documents\Arduino\libraries\LiquidMenu\src\LiquidMenu.cpp:30:0:
C:\Users\Uri\Documents\Arduino\libraries\LiquidMenu\src\LiquidMenu.h:54:123: note: #pragma message: LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.
 # pragma message ("LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.")
                                                                                                                           ^
In file included from C:\Users\Uri\Documents\Arduino\libraries\LiquidMenu\src\LiquidScreen.cpp:30:0:
C:\Users\Uri\Documents\Arduino\libraries\LiquidMenu\src\LiquidMenu.h:54:123: note: #pragma message: LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.
 # pragma message ("LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.")
                                                                                                                           ^
In file included from C:\Users\Uri\Documents\Arduino\libraries\LiquidMenu\src\LiquidSystem.cpp:30:0:
C:\Users\Uri\Documents\Arduino\libraries\LiquidMenu\src\LiquidMenu.h:54:123: note: #pragma message: LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.
 # pragma message ("LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.")
                                                                                                                           ^
In file included from C:\Users\Uri\Documents\Arduino\libraries\LiquidMenu\src\LiquidLine.cpp:30:0:
C:\Users\Uri\Documents\Arduino\libraries\LiquidMenu\src\LiquidMenu.h:54:123: note: #pragma message: LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.
 # pragma message ("LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.")
                                                                                                                           ^
Sketch uses 3134 bytes (9%) of program storage space. Maximum is 32256 bytes.
Global variables use 263 bytes (12%) of dynamic memory, leaving 1785 bytes for local variables. Maximum is 2048 bytes.
```

I think we should display them differently, so it won't confuse users.

urish, 2021-05-24:
Fixed!

These message won't display as long as the program compiles correctly.

closed 2021-05-24 by urish