wokwi/wokwi-features #97: Add resistors, capacitors, transistors...

fred-76, 2021-05-24:
Hello

It should be nice to sort the parts by family in the "+ / add new part" menu. And add
- resistors
- capacitors
- transistors
- diodes
- optocouplers
- special components like PCF8574
Etc.

To date the number of components is very limited.

Cheers

Fred

urish, 2021-05-24:
Thanks for the feedback Fred!

Do you also need simulation for these components? Or just the graphics (so you can add them to the diagram as a reference for someone building the physical circuit)?

fred-76, 2021-05-24:
It depends if Wokwi is aimed to be a simulator like TinkerCAD, being able to warn in the case you overload a component, or what you try simulate doesn't work, or just a graphic tool.

urish, 2021-05-24:
At the moment the aim is to create a digital simulator. Capacitors/transistors/diodes/etc. are analog parts, so they fall out of scope.

There's however a proof-of-concept integration between falstad's CircuitJS simulator and AVR8js (that's the AVR simulation library from Wokwi): https://www.falstad.com/circuit/avr8js/

fred-76, 2021-05-24:
Well, if you don't take some passive components into account, how can we simulate multiple analog buttons ?

For exemple this simulation works in TinkerCad (with 2 lines non I2C LCD), but not on Wokwi:
[image]

[image]

urish, 2021-05-29:
> how can we simulate multiple analog buttons ?

Unfortunately this is not something we support at the moment. While there might be some hacky way to implement this, I suspect we'd need to implement full analog simulation to support this use case.

urish, 2022-03-26:
Closing as some of the use cases are covered by other issues (e.g. resistor divider, mosfet transistors). Capacitors would probably need full analog simulation, which is not planned any time soon.

closed 2022-03-26 by urish