wokwi/wokwi-features #100: Virtual file manager with folders and details

goodoomens, 2021-06-04:
It would be very helpful to organize projects with the following features:

- Create, rename, delete folders
- Move projects to folders, out of folders
- Having detailed information such as last opened, last edited, last added
- Optional preview pictures (they help identify projects but I think they should be an optional view)
- Adding tags to projects and filter for them could be useful too
- A "Create new project" button should be part of this UI

Cheers :)

urish, 2021-06-05:
As a first step, we now have a "Create new project" button:

[image]

urish, 2021-09-17:
Right now, I'm leaning more towards "collections" instead of folders. So you could organize projects (either yours or other's public projects) into a collection, in a way similar to YouTube playlists. 

I think this gives you more flexibility compared to folders:
- You are not limited to just your projects
- You can share your collections with others. E.g. a "Great LED Matrix Arcade Games" collection ;-)
- A single project can live in multiple collections

Would love to hear thoughts - is this a bad idea? What advantages do classic "folders" have over collections?

terrypin999, 2021-10-13:
As discussed briefly with you on Discord yesterday (12th Oct 2021) I hope you'll please quickly fix what i (and others I've been in touch with) regard as a serious flaw. Namely that after changing a sketch name from the default generic 'sketch.ino' to a **unique** and meaningful name (e.g. 'SeveralThings2LEDs-terrypin.ino'), the name remains 'sketch.ino' when you later return to the main window.

P.S. I appreciate that you need some way to resource and prioritise feature requests, but I seriously question whether asking the many users like me that love your program to **PAY** to up-vote them gives you a realistic assessment of demand!

Terry

DiskDoctor, 2021-12-25:
Good idea. How do you find other people's projects that are 'listed' ?

urish, 2022-05-11:
This feature request has almost made it to the [top of the list](https://wokwi.com/features), so I'm starting to plan the implementation. My thoughts so far:

There will be two kinds of collections:
  1. System/global collections - e.g. Raspberry Pi Pico projects, MicroPython projects, Rust, Servo projects, FastLED projects, etc 
  2. User-curated collections - See some specific thoughts in [this comment](https://github.com/wokwi/wokwi-features/issues/100#issuecomment-922047746).

Open questions:

1. What should be the resolution of the system generated collections?  
   I can think of at several different ways to categorize projects: 
   1. By chip family (ESP32, AVR, STM32, RP2)
   3. By chip (e.g. ATMEGA2560, ATtiny85, ESP32, ESP32-S2, etc)
   4. By board (e.g. Arduino Uno, Arduino Nano, etc.)
   5. By programming language / toolchain (e.g. Arduino Core, MicroPython, CircuitPython, Rust, esp-idf, Pi Pico SDK, perhaps in the future also PlatformIO, Zephyr Project, etc.)
   6. By parts used (e.g. Servo, Stepper Motor, ILI9341, etc.)
   7. By libraries used (e.g. FastLED, RTCLib, etc.) - at least for Arduino Core projects

2. What would the URL structure be like?
    Some proposals for system collections:
    - `wokwi.com/browse/micropython`
    - `wokwi.com/categories/micropython`
    - `wokwi.com/collections/micropython`
    - `wokwi.com/projects/micropython` (but we already use `wokwi.com/projects` for user projects, so we've to be careful here)
    - `wokwi.com/micropython` (that would work if we only had a small amount of system collections)
    
    For user collections:
    - `wokwi.com/makers/<username>/collections/<id>` (which also means we'll have to roll out [maker profiles](https://wokwi.com/makers) to all the users). The `<id>` can be either numeric, or derived from the user defined collection name (e.g. `awesome-led-projects`)
    - `wokwi.com/collections/<id>` - In this case, since `<id>` is global (not user specific), we'd probably just use numeric ids as several users can have the same collection name.

Would love to hear what others think.

leftCoast, 2022-05-11:
I would have no system generated collections. Pretty quickly people will just have to find workarounds to undo them because they sort stuff in ways no one could imagine.

urish, 2022-05-11:
> Pretty quickly people will just have to find workarounds to undo them

What do you mean by "undoing them"?

DiskDoctor, 2022-05-11:
You can always consider 2 levels of organization. 
1. Collections
2. Tags/Categories

This could allow searching across collections while retaining some sense of organization

Can we also have a Public v Private option?

Thanks!

leftCoast, 2022-05-13:
What I mean by "undoing them" Is that at the start, there is pretty much no way to know how this will be used. Every one will probably have a completely different idea how to sort their stuff. I'm seeing folders here. In this case any sort of automatic sorting may very well just get in the way of whatever the user has in mind. Possibly, once this has matured and common patterns begin so resolve, some sort of automatic sorting may turn out to be a good addition.

leftCoast, 2022-08-15:
Ok, now I have a use case for this. The plan is to create a set of example sketches for the LC_baseTools library. There is currently about three people interested in working on this (including myself) So three people should be able to edit the simulations but everyone should be able to run and share them.

Also, if this works. I can see it being a very popular way to serve out example code. Unlike a video, you will know if it works or not. And you can tweak it without dealing with getting hardware to run for you.

I thought about doing this through github, but I far rather hack away at small stuff on Wokwi itself. And the back and forth of editing and changes can be very fast between people doing Wokwi. So I can't see how a shared github could help.

Possibly we'd need some sort of lock mechanism for setting who's editing at this moment? And possibly something to tell the current running users that there is now a fresh copy to see.

Twistx77, 2022-11-24:
Hi, wouldn't it ever to just use something like the VSCode plugin and use already all the organization that the OS and VSCode allow?

The pluggin is already working in alfa. Maybe it makes more sense to vote for it rather than this? [https://github.com/wokwi/wokwi-features/issues/414](https://github.com/wokwi/wokwi-features/issues/414)

apollolabsdev, 2023-08-02:
> This feature request has almost made it to the [top of the list](https://wokwi.com/features), so I'm starting to plan the implementation. My thoughts so far:
> 
> There will be two kinds of collections:
> 
> 1. System/global collections - e.g. Raspberry Pi Pico projects, MicroPython projects, Rust, Servo projects, FastLED projects, etc
> 2. User-curated collections - See some specific thoughts in [this comment](https://github.com/wokwi/wokwi-features/issues/100#issuecomment-922047746).
> 
> Open questions:
> 
> 1. What should be the resolution of the system generated collections?
>    I can think of at several different ways to categorize projects:
>    
>    1. By chip family (ESP32, AVR, STM32, RP2)
>    2. By chip (e.g. ATMEGA2560, ATtiny85, ESP32, ESP32-S2, etc)
>    3. By board (e.g. Arduino Uno, Arduino Nano, etc.)
>    4. By programming language / toolchain (e.g. Arduino Core, MicroPython, CircuitPython, Rust, esp-idf, Pi Pico SDK, perhaps in the future also PlatformIO, Zephyr Project, etc.)
>    5. By parts used (e.g. Servo, Stepper Motor, ILI9341, etc.)
>    6. By libraries used (e.g. FastLED, RTCLib, etc.) - at least for Arduino Core projects
> 2. What would the URL structure be like?
>    Some proposals for system collections:
>    
>    * `wokwi.com/browse/micropython`
>    * `wokwi.com/categories/micropython`
>    * `wokwi.com/collections/micropython`
>    * `wokwi.com/projects/micropython` (but we already use `wokwi.com/projects` for user projects, so we've to be careful here)
>    * `wokwi.com/micropython` (that would work if we only had a small amount of system collections)
>    
>    For user collections:
>    
>    * `wokwi.com/makers/<username>/collections/<id>` (which also means we'll have to roll out [maker profiles](https://wokwi.com/makers) to all the users). The `<id>` can be either numeric, or derived from the user defined collection name (e.g. `awesome-led-projects`)
>    * `wokwi.com/collections/<id>` - In this case, since `<id>` is global (not user specific), we'd probably just use numeric ids as several users can have the same collection name.
> 
> Would love to hear what others think.

I'm increasingly becoming in dire need to organize projects. As I pile up more examples, I'm increasingly finding it hard to navigate myself. It would be nice to get this feature sometime soon :) especially as Wokwi is growing in popularity and the number of projects is increasing.

For me, the resolution for the system generated collections probably makes sense by language.  I would also go for the  `wokwi.com/collections/micropython` URL structure. As for user collections, it would be nice to have maker profiles. Sort of a way to showcase someone's work on Wokwi.

urish, 2023-08-02:
Thanks for the feedback!

Some of this is already implemented:
1. For rust and micropython, we have https://wokwi.com/rust and https://wokwi.com/micropython (more pages coming soon - maybe even this week - for esp32, pi-pico, arduino, and stm32)
2. Maker profilers - you can create one at https://wokwi.com/dashboard/profile. Select a username, and you'll be listed at https://wokwi.com/makers