wokwi/wokwi-features #101: Can you add LC_baseTools to wokwi?

leftCoast, 2021-06-06:
Here is the link to its github. There is a released version in the Arduino IDE. I ask because I've been asked a user of yours to add it.

Thanks!

-jim lee

urish, 2021-06-06:
Hi Jim lee, thanks for the suggestion. Library added!

To quickly test, I uploaded the auto_blinkers example to the simulator: https://wokwi.com/arduino/projects/300639412311556616

Does it work as intended?

leftCoast, 2021-06-06:
Yeah, works great! I'm impressed! Now, lets get something like a teensy 3.2 & maybe an Adafruit 1947 touch screen so I can really do something. (I can port my handheld to this then)

Good work! 

-jim lee

closed 2021-06-06 by leftCoast

urish, 2021-06-06:
Thank you :)

Teensy is unlikely to happen any time soon, but there's a [Raspberry Pi Pico](https://docs.wokwi.com/parts/wokwi-pi-pico) cooking which may be useful for you.

But the touch screen sounds like something we can do...

What handheld are you working on?

leftCoast, 2021-06-07:
I run a teensy 3.2 wired to a Adafruit 1947 touch screen with a 3D printed case. I use this for a bunch of stuff. Mostly I code on the teensy 3.2s. I have an entire touch screen OS etc. That I wrote for this setup that I use for making stuff.
For example : [My Arduino Cell phone](http://www.leftcoast.biz/iWeb/Left_Coast/Home_made_cellpone.html)
And this : [My handheld](http://www.leftcoast.biz/iWeb/Left_Coast/The_Left_Coast_Handheld.html)
I can't run the OS stuff on the Uno because its just too small. Is porting processors pretty tough? I'm thinking it probably is because the Teesny comes with its own version of the IDE so there's probably a lot of bookkeeping that comes along with changing the processor.