wokwi/wokwi-features #108: Can't seem to add more than 2 LEDs and only seem to be able to run one pwm.

leftCoast, 2021-06-09:
Here's a little program that mimics fireflies. I wrote it for someone on the Arduino board. Seems that it works reat on an UNO but I'm having no end of troubles running it on wokwi
A) Can't put up more than 2 LEDs
B) Only one will light up.
```Arduino
#include <timeObj.h>
#include <mapper.h>
#include <idlers.h>
#include <blinker.h>

#define CHECK_MS        10       // How often we look. Must be small number.
#define CHANCE_TO_LIGHT 2        // We have a chance of lighting ever "CHANCE_TO_LIGHT"
#define OUT_OF          5000     // out of "OUT_OF" times we check.
#define FADE_MS         200      // We fade in and out about this many Ms.
#define FADE_DITHER     50       // How much in ms +/- to randomise fade.
#define LITE_MS_LOW     100      // Ms
#define LITE_MS_HIGH    1000     // Ms

mapper   fadeInMapper(0,FADE_MS,0,255);
mapper   fadeOutMapper(0,FADE_MS,255,0);

class firefly :   public idler {

   public:
            enum  flyStates { liteOff, fadeIn, liteOn, fadeOut };
            
            firefly(void);
            ~firefly(void);

            void  begin(int inPinNum);
   virtual  void  idle();

            flyStates   flyState;
            int         pinNum;
            int         fadeInMs;
            int         fadeOutMs;
            int         holdlitMs;
            timeObj     checkTimer;
            int         lightTime;
};

firefly::firefly(void) { flyState = liteOff; }

firefly::~firefly(void) { }

void firefly::begin(int inPinNum) {

   pinNum = inPinNum;
   pinMode(pinNum,OUTPUT);
   digitalWrite(pinNum,LOW);
   checkTimer.setTime(CHECK_MS);
   hookup();  
}

void firefly::idle(void) {

   int lightValue;
   
   if (checkTimer.ding()) {
      checkTimer.start();
      switch (flyState) {
         case liteOff   :
            if (random(0,OUT_OF)<=CHANCE_TO_LIGHT) {
               lightTime = 0;
               fadeInMs = random(FADE_MS-50,FADE_MS+50);
               fadeOutMs = random(FADE_MS-50,FADE_MS+50);
               holdlitMs = random(LITE_MS_LOW,LITE_MS_HIGH);
               flyState = fadeIn;
            }
         break;
         case fadeIn    :
            if (lightTime<=fadeInMs) {
               lightValue = fadeInMapper.map(lightTime);
               analogWrite(pinNum,lightValue);
               lightTime = lightTime + CHECK_MS;
            } else {
               lightTime = 0;
               flyState = liteOn;
            }
            break;
         case liteOn    :
            if (lightTime<=holdlitMs) {
               lightTime = lightTime + CHECK_MS;
            } else {
               lightTime = 0;
               flyState = fadeOut;
            }
            break;
         case fadeOut   :
            if (lightTime<=fadeOutMs) {
               lightValue = fadeOutMapper.map(lightTime);
               analogWrite(pinNum,lightValue);
               lightTime = lightTime + CHECK_MS;
            } else {
               lightTime = 0;
               analogWrite(pinNum,0);
               flyState = liteOff;
            }
         break;
      }
   }
}

// *************************************************
// THE MAIN PROGRAM
// *************************************************

firefly fireflyArray[6];
blinker  heartbeat;      // Flashes pin 13 to show everything is still alive.

void setup(void) {

   fireflyArray[0].begin(3);
   fireflyArray[1].begin(5);
   fireflyArray[2].begin(6);
   fireflyArray[3].begin(9);
   fireflyArray[4].begin(10);
   fireflyArray[5].begin(11);
   heartbeat.setOnOff(true);
}

void loop(void) { idle(); }
```
-jim lee

sutaburosu, 2021-06-09:
Can you please share a link to this sketch in Wokwi.

leftCoast, 2021-06-09:
Well I take back the bit about only one flashes. I saw the other one flash once.

[The link](https://wokwi.com/arduino/new?template=arduino-uno)

leftCoast, 2021-06-09:
AH! I see the 3rd LED is below another. I never saw it.

Sorry to be a bother.

closed 2021-06-09 by leftCoast

sutaburosu, 2021-06-09:
I was confused by those includes that I didn't recognise. I thought we didn't have the whole code. Sorry about that.

[Here's that sketch with 12 LEDs and a slower fade rate](https://wokwi.com/arduino/projects/300894426757071368).  The LEDs seem to do what I would expect.  Oh, you just closed this.  It's working for you now then.  :champagne:

leftCoast, 2021-06-09:
Yeah, the issue really was that when I added the third and subsequent LEDs, I couldn't see them as they were spawned under one of the others.

leftCoast, 2021-06-11:
Yes, anything but hide it. :)

leftCoast, 2021-06-16:
I noticed the offset. This is good, I'd offset about 2x as much. But still, this should stop people like me from getting mixed up.

One other thing.. Is it ok to add more to this thread?

Take the zoom of size off the editor dragging, and add it to some sort of zoom only screen control. Even a "type in the percent" kinda' thing would be a vast improvement. This way your dragging up,dn, side to side will work **way** better and you won't feel like you suddenly jumped into warp drive away from your project. I've totally lost my project when trying to drag it over a little bit. Its like it zoomed out so far it became a lost dot off the screen somewhere.

And, while you're at it, set up an area for a tool menu and get the zoom square off the working panel. I keep clicking it when I want to click buttons. (It gets in the way)

I only tell you this stuff because your emulator actually works the best I've seen and I like using it. In fact, it works great for posting example code on forums etc.

urish, 2021-06-17:
Thanks for the feedback Jim! 

Mind sharing a quick sketch of what you have in mind? I feel like visuals will help me better understand :)

leftCoast, 2021-06-19:
Before I do that.. Something broke and now I can't edit anything in the text window. (Mac, safari) maybe it was when you fixed the resistor thing for me?

leftCoast, 2021-06-19:
l
[image]

urish, 2021-06-19:
Thanks Jim! Now it's much more clear to me :)

sutaburosu, 2021-06-19:
If I remember correctly, everyone who has commented on the overall UI design wants to get rid of the full-screen button.

If your metrics show that I'm the only one who ever presses this button, can I please at least have a keyboard shortcut?  F11 is taken, but my fingers learned many years ago that Alt+Enter toggles full-screen in games and some other DirectX/OpenGL apps.

Pop-up scroll-bars are a great idea.  If the bars work conventionally (e.g. are scaled to represent how much of the active workspace is visible on-screen) there would no need for the magnification popup.

urish, 2021-06-19:
> Get rid of the full-screen button

I remember someone suggesting to replace its functionality with "zoom to fit"

> Alt+Enter

Great idea. You got it!

btw. the + / - key already work for zooming in/out.

> there would no need for the magnification popup.

One option I've been thinking about is similar to what you have in google slides:

[image]

And then add "Full screen" as an option there. Thoughts?

sutaburosu, 2021-06-19:
Yes, I like the idea of that menu in Wokwi. Perhaps also "Zoom lock" might be useful for touch-interface users? With a padlock icon where the full-screen one used to be.

urish, 2021-06-19:
> With a padlock icon where the full-screen one used to be.

Do you happen to have an example of an app with this kind of interface? Unfortunately, I'm not a heavy mobile user, so mobile UX is always an afterthought for me

leftCoast, 2021-06-20:
The menu looks great! My main issue is to get the zooming away from the mouse touch, or finger touch. I just showed the popup because I'd seen it done that way. Serving suggestion kind of thing. Anyway that's easy to code would probably be fine.

I might even drop the scroll bars and just drag the screen around **if** the zoom function was gone.

urish, 2021-06-27:
Mind opening a new issue for the drop-down zoom menu? So we won't lose it between all these comments :)

leftCoast, 2021-06-27:
Certainly!