wokwi/wokwi-features #109: OLED_I2C

oorazgeldiyeva, 2021-06-09:
Hello, @urish !
Could you add OLED display on I2C module and is library OLED_I2C?

Thank you in advance!!

Best regards,
oorazgeldiyeva

urish, 2021-06-09:
Hello, we do have an OLED on I2C module, that's the SSD1306:

https://wokwi.com/arduino/projects/296836708191699465

Is this the model that you need or do you need another kind of I2C OLED display?

oorazgeldiyeva, 2021-06-09:
I need another kind of I2C OLED display, which have 4 pins: VCC, GND, SDA, SCL. 
But if it ie not possible to add it, I will try to work with another one  you mentioned.

Thank you for reply!

urish, 2021-06-09:
The question is what model of chip your OLED display has. 

I believe that SSD1306 is the most popular, and you can run it with just 4 pins:

connect VCC to VIN, GND to GND, SDA to Data and SCL to Clk and leave the other pins (DC / RST / CS / 3V3) unconnected

urish, 2021-06-10:
duplicate: #106

closed 2021-06-10 by urish