wokwi/wokwi-features #110: request

pol-122, 2021-06-09:
Hello!
I need DALLAS DS18B20, it is temperature chip. Could you add it? And if it is possible its library DallasTemperature.h too

urish, 2021-06-09:
Library added! I'll look into adding the sensor  itself over the next couple of days.

What project are you working on?

pol-122, 2021-06-10:
Oh, thank you!
It is just a simple project, which will show temperature on oled or lcd display.

If I know, there is not OLED_I2C library. So maybe I will use LCD I2C.

One more request. Could you add stepper? Its library exist, but sensor itself not

urish, 2021-06-12:
> One more request. Could you add stepper? Its library exist, but sensor itself not

Mind opening a new issue for the stepper?

urish, 2021-06-25:
> Library added! I'll look into adding the sensor itself over the next couple of days.

It'll take a while until I get to implementing OneWire simulation...

urish, 2021-11-03:
Hey @AleksandraStecka! 

It's currently [open for votes](https://wokwi.com/features#feature-916171688). The more votes it gets, the sooner we can get to work on it.

Are there any other parts you need for your uni project?

flixmyswitch, 2022-01-09:
> 

Hi have been looking through the sensor but cant see this, so not sure if it has been created.  I see if way down the pecking order in the votes, sad face

urish, 2022-01-10:
@flixmyswitch not yet. It did get 2 votes, which is a good starting point :)

urish, 2022-01-12:
Good pointer from facebook discussion:  https://github.com/orgua/OneWireHub

nickinit, 2022-06-28:
Hello! I also really need the ability to simulate a DS18B20 sensor. This is a very useful addition in my opinion!

urish, 2022-12-14:
@bonnyr started working on [an implementation](https://github.com/bonnyr/wokwi-ds1820-custom-chip/) using the Custom Chip API.

bonnyr, 2022-12-14:
Implementation (first pass, intended to be generalised when I have time)

Repo for source is here: https://github.com/bonnyr/wokwi-ds1820-custom-chip/tree/main/example
The `README.md` should point to an example project I used for testing (https://wokwi.com/projects/349898396478210642)

This makes use of @urish addition of pulling custom chips from GitHub (kudos :) )

urish, 2023-02-26:
Now available in the purple Add Part menu, as a beta feature:

[image]

closed 2023-02-26 by urish