wokwi/wokwi-features #116: Export project

arcostasi, 2021-06-14:
Add an option to export the project files into a zip file. I think it can be done in the Save pop-up menu just below the Save a copy.

urish, 2021-06-17:
Done!

I added it to the file drop down menu (where you can create new files, rename, delete, etc.), but I'm not sure it's the best place - should we move it to the save popup menu?

[image]

arcostasi, 2021-06-17:
I believe so @urish, maybe beginners or even visitors don't find this feature easy in projects.

urish, 2021-06-19:
How about now?

[image]

arcostasi, 2021-06-19:
Perfect @urish, I believe will give users more confidence in having a save option. Thank you very much

closed 2021-06-19 by urish