wokwi/wokwi-features #117: left scroll is not working as expected

matititam, 2021-06-17:
project link: https://wokwi.com/arduino/projects/301543301987697165

lcd.scrollDisplayLeft();

[image]

urish, 2021-06-17:
Thanks for reporting!

Can you please attach a reference video of how the same code works on the physical hardware?

matititam, 2021-06-20:
I do not have a 2004 display. I have 1602 displays with me. the issue is there with 1602 as well. I will check it on real hardware and share the reference by end of this week.

urish, 2021-06-20:
Thank you!

Also, if you have an example of a 1602 code that behaves differently in the sim v.s. real life that would be helpful!

matititam, 2021-06-30:
Link for **Right** scroll example: https://wokwi.com/arduino/projects/302803492608672266
how it looks on the simulator (below image)
[image]

How it works on the actual hardware: (link to the video)
Scrolling the text right - lcdsetCursor(0,1) to bring the text one row down (https://drive.google.com/file/d/1T5wpCTJ9QczeRr2UA6UsaSspVN9cBZe1/view?usp=sharing)

---------
Scrolling the text left: (https://drive.google.com/file/d/1ca0ADUZPBbO-d-dswR4W0QOdlRYKUXQD/view?usp=sharing)

if not, the text will scroll in the first row

urish, 2021-07-03:
Alright, fixed!

Thanks for the detailed bug report :)

closed 2021-07-03 by urish