wokwi/wokwi-features #120: sliding potentiometer is not easy to operate

matititam, 2021-06-20:
link: https://wokwi.com/arduino/projects/301206238041997832

I am not able to move the potentiometers easily. More or less it is impossible sometimes. here is a small video.
[image]

urish, 2021-06-20:
Seems to work well here (Chrome) - which web browser are you using? 

If you cancel the rotation of the potentiometer (currently it's 90 degrees in this project), do they work better?

matititam, 2021-06-20:
I am using Firefox. Even after rotation, the UI will fail after a couple of clicks. they will stop responding completely. 

I checked with Chrome. It works well with chrome.

urish, 2021-06-20:
I can confirm this reproduces on FireFox. Moving to wokwi-elements.