wokwi/wokwi-features #122: Can we add the Adafruit 1947 cap touch screen?

leftCoast, 2021-06-20:
Not sure about others, but I use the 1947 a **lot** for my projects. It has the smallest form factor for scree size, pretty good SD card placement, and fits in a handheld well. ( I never actually use it for its intended purpose as a UNO Shield. )

[For example here's my 3D case designed for this screen](https://www.thingiverse.com/thing:4706082)

urish, 2021-06-25:
* TFT Controller: [ILI9341](https://cdn-shop.adafruit.com/datasheets/ILI9341.pdf)
* Touch controller:
  * Capacitive varation: [FT6206 (I2C)](https://cdn-shop.adafruit.com/datasheets/FT6x06+Datasheet_V0.1_Preliminary_20120723.pdf).
  * Resistive version: [STMPE610 (SPI)](http://www.adafruit.com/datasheets/STMPE610.pdf)
* SD Card interface

Overall, this seems like a pretty complex part. We'd first need to implement SD Card support, and that may take a while. Unless you don't really use the SD Card in your projects?

leftCoast, 2021-06-26:
I use the SD card in all of them. And the 1947 is SPI with Capacitive.

urish, 2021-06-27:
So I guess SD Card simulation is also important for you?

leftCoast, 2021-06-27:
Yeah, my stuff tends to use a lot of icons and things like that. (And data files) So an SD card is a kinda' high priority.

urish, 2021-08-21:
Update: there's a prototype of the TFT -

https://wokwi.com/arduino/projects/307567201804616256

Not very polished, but functional and ready for feedback :)

urish, 2021-09-12:
TFT is ready! https://docs.wokwi.com/parts/wokwi-ili9341

closed 2022-01-27 by urish